jketterl/openwebrx #272: Unable to connect to RTL-SDR dongle after latest git pull

ofadam, 2021-09-30:
**Describe the bug**
After updating openwebrx on 9/29, I'm no longer able to get the previously working installation to load an RTL-SDR dongle and stay alive.

**Installation method**
Installed from Ubuntu packages

**Versions**
The latest of openwebrx on Ubuntu 21

**Log messages**
2021-09-30 04:07:00,214 - owrx.connection - DEBUG - client connection initialized
2021-09-30 04:07:00,244 - owrx.source - INFO - Started sdr source: rtl_connector -P 0 -g 35.0 -s 2048000 -f 144900000 -p 54397 -c 56493
setting up control socket...
control socket started on 56493
socket setup complete, waiting for connections
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

client connection established
2021-09-30 04:07:00,348 - owrx.source.connector - DEBUG - opening control socket...
control connection established
2021-09-30 04:07:00,370 - owrx.dsp - DEBUG - wiring new output of type audio
Found Rafael Micro R820T tuner
2021-09-30 04:07:00,867 - owrx.dsp - DEBUG - wiring new output of type smeter
2021-09-30 04:07:00,869 - owrx.dsp - DEBUG - wiring new output of type meta
2021-09-30 04:07:00,870 - owrx.dsp - DEBUG - wiring new output of type secondary_fft
2021-09-30 04:07:00,870 - owrx.dsp - DEBUG - wiring new output of type secondary_demod
2021-09-30 04:07:00,871 - owrx.dsp - DEBUG - received STATE_RUNNING, attempting DspSource restart
Allocating 2 zero-copy buffers
2021-09-30 04:07:00,965 - owrx.bands - DEBUG - reloading bands from disk due to file modification
2021-09-30 04:07:00,979 - owrx.bookmarks - DEBUG - reloading bookmarks from disk due to file modification
closing client socket
2021-09-30 04:07:01,329 - owrx.fft - DEBUG - Spectrum thread initialized successfully.
double free or corruption (out)
control connection ended
Aborted (core dumped)

jketterl, 2021-09-30:
First of all, please try to be more precise on the version information, I don't like guessing since that usually takes me down the wrong road.

"The latest of openwebrx" is not very specific. The latest what? I'm going to assume you mean the latest version from the experimental repository, which would be `1.2.0~921` right now. Please check if that is correct. Please also double-check the versions of the libraries: csdr `0.18.0~135`, pycsdr `0.18.0~14`, digiham `0.6.0~116` and pydigiham `0.6.0~4`. If any of these is older, please update and retry. Also note that these versions will change with time, so it's important to give a more stable reference that I can work with.

"Ubuntu 21" is not complete either, It's either Ubuntu 21.04 or "Hirsute". Luckily, this is not ambiguous right now, but it will be in a a few weeks.

Second, I was not able to reproduce this bug. I have set up a fresh Ubuntu 21.04 virtual machine, and installed OpenWebRX from the experimental repository on it, and it worked as expected (with an RTL-SDR).

Third, in order to be able to debug this, I will need more information. Ultimately, I need to see a stack trace of the problem. This can be obtained by running OpenWebRX inside a debugger like `gdb`, or by analyzing the core that was dumped with said debugger. The core dump can be analyzed on a third party machine, but only if the packages are exactly the same (which means that this method only works if the machine is fully updated and there's no new packages until the analysis can be performed).

ofadam, 2021-09-30:
Understood on the lack of specifics – I'll do better with any future tickets.

It looks like it was due to python3-csdr and python3-digiham taking more time to become visible as updates via apt – they appeared this morning and all is working great again. Thanks for the help.

closed 2021-09-30 by ofadam