jketterl/openwebrx #276: ADS-B decoder

AsciiWolf, 2021-11-02:
**Feature description**
From [faa.gov](https://www.faa.gov/nextgen/equipadsb/capabilities/ins_outs/): _ADS-B Out works by broadcasting information about an aircraft's GPS location, altitude, ground speed and other data to ground stations and other aircraft, once per second. Air traffic controllers and aircraft equipped with ADS-B In can immediately receive this information. This offers more precise tracking of aircraft compared to radar technology, which sweeps for position information every 5 to 12 seconds._
ADS-B decoder should be fairly simple to implement and would allow ADS-B data to be decoded directly using OpenWebRX.

**Target audience**
Many radio hobbyists and plane spotters love decoding ADS-B data. It would be great if OpenWebRX supported this. :-)

jketterl, 2021-11-21:
There is a bunch of ADS-B decoders out there; for this to work, however, I need one that can accept IQ data in some form, and does not directly access the SDR device. The only one I have worked with personally is dump1090, which does not match those requirements.

jketterl, 2023-09-12:
Alright, for those who have missed it... this was implemented in the latest development version. Feel free to test.