jketterl/openwebrx #280: Raspberry Pi Image udev rule for IIO devices broken

g4wsz, 2021-11-27:
**Describe the bug**
I think that the udev rule that would detect PlutoSDR devices is broken in the Raspberry PI SD image, so that those devices aren't usable by non-root users.

**To Reproduce**

Boot Pi image, plug in Pluto USB cable then run SoapySDRUtil as a normal user then as root (with sudo).

**Expected behavior**

The Pluto device should appear to normal users but doesn't, unless "sudo" is used.

**Installation method**

Downloaded and unpacked SD card image fro Pi, wrote to SD card. Booted on a Raspberry Pi4 (2GB).

**Versions**

2021-08-03-OpenWebRX-full.img

**Log messages**

...   owrx.source - INFO - Started sdr source: soapy_connector -s 2400000 -f 433000000 -p 33905 -c 34011 -d driver=plutosdr
...  owrx.connection - WARNING - SDR device "plutosdr" has failed, selecting new device

**Additional context**
The /etc/udev/rules.d/90-libiio.rules file contains "junk" instead of the path to the iio_info binary (eg /usr/bin/) so I suspect some variable evaluation has gone wrong. Unfortunately, for once in my life I didn't bother to keep the original file on the SD card but I will try to fish that out ASAP.

jketterl, 2021-11-27:
You're right... the file was probably taken from here, but it doesn't make sense if it doesn't pass through cmake as intended. This probably never worked (but it was in place since October '20...).

I'll need to check if i can activate the corresponding cmake option...

g4wsz, 2021-11-27:
The contents as distributed on the SD card image are:
`SUBSYSTEM=="usb", PROGRAM=="/bin/sh -c '@CMAKE_INSTALL_FULL_BINDIR@/iio_info -s | grep %s{idVendor}:%s{idProduct}'", RESULT!="", MODE="666"`

jketterl, 2021-11-27:
OK seems like the solution is to simply remove the manually added file. The new packages for Debian Bullseye seem to have added this file, and since the next release image will be based on Bullseye this should be fixed by simply undoing the current solution (which is probably nothing more than a workaround for Debian Buster).

g4wsz, 2021-11-27:
Thanks great and thanks for the rapid response(s)
:-)

jketterl, 2021-11-27:
I uploaded an experimental image at https://www.openwebrx.de/download/rpi.php?path=experimental/bullseye/ - I'd appreciate any feedback if you can try it. It contains an experimental version of OpenWebRX, and some software is not up to date due to compatibility reasons (most notably, WSJT-X and JS8Call), so I'd recommend a separate SD card for testing this.

g4wsz, 2021-11-28:
On 27/11/2021 18:34, Jakob Ketterl wrote:
> I uploaded an experimental image at
> https://www.openwebrx.de/download/rpi.php?path=experimental/bullseye/
> <https://www.openwebrx.de/download/rpi.php?path=experimental/bullseye/>
> - I'd appreciate [...]
That certainly resolves the IIO rules file issue Jakob, I see it's part 
of libiio0 in your bullseye-based image.

Adding the Pluto driver worked smoothly!

73
Steve

jketterl, 2022-06-17:
This is now in the latest stable RPi images.

closed 2022-06-17 by jketterl