jketterl/openwebrx #282: Longitue zero seems a problem

g4wsz, 2021-11-29:
**Describe the bug**
I can't enter my station location as 52N, 0E ... it accepts 52 but not 0.
The longitude reverts to it's default/previous value.

**To Reproduce**
Try entering 0 or 0.0 as the longitude, then Apply.

**Expected behavior**
I expect the longitude to show as 0!

**Installation method**
Raspberry Pi SD card image (latest)

**Versions**
What version of OpenWebRX are you running? (Check on startup, or see `owrx/version.py`. If a `-dev` version is used, ideally state the commit the issue is appearing on)

**Log messages**
Are there any relevant messages relating to the bug in the output / log of OpenWebRX? (On most installations, the log should be available using the command `sudo journalctl -u openwebrx`)

**Additional context**
Add any other context about the problem here.

g4wsz, 2021-11-29:
Yes, indeed, I found 0.0001 worked fine ;-)  I suppose I took note of a suggestion somewhere not to enter your exact location, so I thought entering (52,0) would make it clear that it was only an approximate position. Thanks for your efforts!

jketterl, 2021-12-14:
Alright, I have been able to reproduce this, and as it turns out, the error was not really of the nature I was expecting. As it turns out, this is actually a problem with the `location-picker` widget that is used to display the little map snippet. A latitude / longitude of `0` is not accepted and instead overridden with its defaults.

I have created a little fix for this problem and sent it upstream, see this pull request: https://github.com/cyphercodes/location-picker/pull/11

I have applied this fix to the current OpenWebRX codebase as well, so this should work as expected for the current development version now.

g4wsz, 2021-12-14:
Glad you pinned it down!  Thanks for the fix and letting us know :-)

jketterl, 2022-06-17:
in stable with the release of 1.2.0

closed 2022-06-17 by jketterl