jketterl/openwebrx #287: RPi3 docker: sdrplay not found

sgofferj, 2022-01-02:
**Describe the bug**
I have a generic msi2500 dongle connected to the RPi. owrx-docker does not find the dongle. The kernel drivers are blacklisted on the host. lsusb on the host shows the device connected. /dev/bus/usb is shared to the container. rtlsdr dongles work fine. On my laptop, the device works fine with soapy/sdr++ with sdrplay settings. 

**To Reproduce**
Try to use an sdrplay-compatible dongle in owrx-docker

**Expected behavior**
The dongle should work in owrx.

**Installation method**
Docker

**Versions**
jketterl/openwebrx:stable@sha256:54ffd2260d04f1f8cc249e2ecf2a55799c21c8b663e7d739f91725219bc50ad5

**Log messages**
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2021-12-30 18:29:02,520 - owrx.__main__ - INFO - OpenWebRX version v1.1.0 starting up...
2021-12-30 18:29:02,539 - owrx.config.migration - DEBUG - migrating config from version 3
2021-12-30 18:29:02,541 - owrx.config.migration - DEBUG - detected waterfall option: GoogleTurboWaterfall
2021-12-30 18:29:03,433 - owrx.source - INFO - Started sdr source: soapy_connector -s 10000000 -g auto -f 435000000 -p 36927 -c 54011 -d driver=sdrplay
setting up control socket...
control socket started on 54011
socket setup complete, waiting for connections
client connection established
2021-12-30 18:29:03,538 - owrx.source.connector - DEBUG - opening control socket...
control connection established
[INFO] devIdx: 0
[INFO] hwVer: 1
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
[INFO] Using format CF32.
libusb: error [submit_iso_transfer] submiturb failed error -1 errno=12
[ERROR] error in activateStream() - Init() failed: sdrplay_api_Fail
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
2021-12-30 18:29:08,672 - owrx.connection - DEBUG - client connection initialized
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
2021-12-30 18:29:11,640 - owrx.dsp - DEBUG - received STATE_RUNNING, attempting DspSource restart
2021-12-30 18:29:11,899 - owrx.bands - DEBUG - reloading bands from disk due to file modification
2021-12-30 18:29:11,918 - owrx.bands - WARNING - Frequency for wspr on 4m is not within band limits: 70091000
2021-12-30 18:29:11,923 - owrx.bands - WARNING - Frequency for q65 on 13cm is not within band limits: 2301065000
2021-12-30 18:29:11,924 - owrx.bands - WARNING - Frequency for q65 on 13cm is not within band limits: 2304065000
2021-12-30 18:29:11,928 - owrx.bookmarks - DEBUG - reloading bookmarks from disk due to file modification
WARNING: SoapySDR::Device::readStream timeout!
2021-12-30 18:29:13,138 - owrx.fft - DEBUG - Spectrum thread initialized successfully.
2021-12-30 18:29:13,139 - csdr - DEBUG - Command = nc -v 127.0.0.1 36927 | csdr fft_cc 4096 2863.68843069874 | csdr logaveragepower_cf -70 4096 388 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
Connection to 127.0.0.1 36927 port [tcp/*] succeeded!
client connection established
WARNING: SoapySDR::Device::readStream timeout!
2021-12-30 18:29:14,272 - owrx.cpu - DEBUG - cpu usage thread starting up
2021-12-30 18:29:14,273 - owrx.connection - DEBUG - handshake complete, handing off to OpenWebRxReceiverClient
2021-12-30 18:29:14,299 - csdr - DEBUG - Command = nc -v 127.0.0.1 36927 | csdr shift_addfast_cc --fifo /tmp/openwebrx/openwebrx_pipe_shift_pipe_281473604842944 | csdr fir_decimate_cc 907 0.000165380374862183 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx/openwebrx_pipe_bpf_pipe_281473604842552 0.029024 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx/openwebrx_pipe_squelch_pipe_281473604842720 --outfifo /tmp/openwebrx/openwebrx_pipe_smeter_pipe_281473604843224 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr fractional_decimator_ff 1.0000325010562843 | csdr deemphasis_nfm_ff 11025 | csdr agc_ff --profile slow --max 3 | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
2021-12-30 18:29:14,315 - owrx.dsp - DEBUG - adding new output of type audio
2021-12-30 18:29:14,320 - owrx.dsp - DEBUG - adding new output of type smeter
Connection to 127.0.0.1 36927 port [tcp/*] succeeded!
AGC PARAMS:
  hang_time = 600
  reference = 0.800000
  attack_rate = 0.010000
  decay_rate = 0.000100
  max_gain = 3.000000
  initial_gain = 1.000000
  attack_wait_time = 0
  gain_filter_alpha = 1.500000
client connection established
csdr shift_addfast_cc: fifo control mode on
csdr fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
csdr fractional_decimator_ff: not using taps
csdr squelch_and_smeter_cc: fifo control mode on
csdr bandpass_fir_fft_cc: fir_decimate_cc: taps_length = 24187
csdr fir_decimate_cc: taps_length = 24187
csdr fir_decimate_cc: padded_taps_length = 24192
csdr fir_decimate_cc: taps = 1c6b9270
csdr fir_decimate_cc: NEON aligned taps = 1c6b9280
fifo control mode on
csdr bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 137) + (input_size = 376) - 1
(overlap_length = 136) = taps_length - 1
csdr shift_addfast_cc: reinitialized to -0
csdr squelch_and_smeter_cc: initial squelch level is 1e-15
csdr bandpass_fir_fft_cc: filter initialized, low_cut = -0.3628, high_cut = 0.3628
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
WARNING: SoapySDR::Device::readStream timeout!
```

jketterl, 2022-01-02:
This is a low-level communication problem between the device and the SDRPlay software, nothing that could be fixed in OpenWebRX.

Since the SDRPlay software only supports official SDRPlay hardware, a "generic msi2500 dongle" is not supported. You'll probably find it pretty hard to get support from SDRPlay, too. I supposed it's time to contact the hardware manufacturer / vendor.

closed 2022-01-02 by jketterl

sgofferj, 2022-01-02:
As I wrote before, the dongle works just fine with soapy on my laptop. It also works on another Pi outside of docker. I believe there might be a problem with the docker image.