jketterl/openwebrx #289: Scroll waterfall left and right like KiwiSDR

bhall7, 2022-01-21:
Hello,

It would be awesome if OpenWebRX would enable scrolling the waterfall left and right like KiwiSDR.

Thanks!

jketterl, 2022-01-21:
If this is a feature request, please use the corresponding issue template.

Duplicate of #97.

Please also see detailed explanation [here](https://groups.io/g/openwebrx/message/937).

Things are certainly not going to move faster by requesting the same things over and over again...

closed 2022-01-21 by jketterl

bhall7, 2022-01-21:
Apologies. I searched for this, but couldn't find anything on panning waterfall left/right. I didn't realize it was referred to as changing the center frequency via the UI. Thanks for sharing the additional information. Hoping this can be overcome in the future. Appreciate all the work done on this project.