jketterl/openwebrx #300: No decoding of FT8 etc... and empty map

loewal, 2022-05-27:
**Describe the bug**
I'm receiving clear signals on 20m of jt8 etc.
There is no decoding and the map contains no red or green blocks.
This worked well for months.

**To Reproduce**
Tune to these frequencies on 20 or 40m

**Expected behavior**
List of transmissions

**Installation method**
Compiled maually

**Versions**
v.1.1.0

**Log messages**
I don't know

**Additional context**
I recompiled js8py, digiham, codec2, owrx_connector, codecserver and csdr.
No change....

loewal, 2022-05-27:
After compilation and restart of Odroid N2+ everything was suddenly ok.....

jketterl, 2022-05-27:
did you check the clock?

loewal, 2022-05-27:
Helas, I didn’t (know that).
Time is ok now.
Thanks!

jketterl, 2022-05-27:
All of the WSJT-X modes use the clock to synchronize transmission intervals, so your clock must be somewhat accurate. The tolerance depends on the individual mode, i think for FT8 the maximum delta is thee seconds. Unfortunately, this isn't immediately obvious since from the point of view of the system, there's no immediate error occurring, it just doesn't find any usable signals.

There's also other bugs in WSJT-X that appear after prolonged operation, which can be fixed by clearing out the temporary directory.

I'm also encountering complete failure of the WSJT-X decoders (segmentation faults) which usually require a restart. I don't know anything about the cause, but as far as I can tell this behavior is not caused by OpenWebRX.

If the decoder itself fails, there should be error messages in the log.

closed 2022-05-27 by jketterl

loewal, 2022-05-27:
Thanks for your explanation Jakob!