jketterl/openwebrx #303: Intermittent audio on Apple i-telephone

eroyee, 2022-06-20:
**Describe the bug**

Exhibits 'stuttering audio on mobile browsers' per https://groups.io/g/openwebrx/topic/91848686 it sounds like intermittent audio dropout.

**To Reproduce**
Use an i-telephone with OpenWebRX and tune to any any station. After ~ 1minute the audio becomes intermittent.

**Expected behavior**
Continual replay of the station audio without issue.

**Installation method**
RPi apt-get per your instructions.

**Versions**
1.2

**Log messages**
Nil applicable.

**Additional context**

Jakob,

I have just been working with one of the people affected with 'stuttering audio' when used with an i-telephone. 

It turns out that disabling adpcm compression for the audio stream appears to resolve the issue, so this bug report is as much about how to workaround the problem until a 'proper' resolution may be achieved. 

Like you I don't have one of these telephones so I'm unable to do a lot more, but I wonder if it the issue could be related to https://github.com/jketterl/openwebrx/issues/203 and the recent introduction of the adpcm option? 

It may be of some note that audio is initially ok, but begins to fail after a short time. It is not reproduceable for me in my environment, and a brief trawl around the internet doesn't appear to show any recent issues with adpcm and appleOS, although it may not be used much elsewhere and I suppose any bug introduced by Apple could take a while to show up.

Regards.

jketterl, 2022-06-20:
As already stated in the linked thread, there is no support for mobile browsers, Apple, or Safari, so I will close this issue without further discussion.

If you want to discuss the ADPCM implementation, please get in touch on a suitable channel.

closed 2022-06-20 by jketterl

bhall7, 2022-06-20:
Sad