jketterl/openwebrx #314: Choppy audio when compressed ADPCM on Safari web browser .

Terminator0007, 2022-09-04:
I'm using only Apple devices ( Mac mini, Macbook, iPad, iPhone, all with newest OS ) but can check on Safari, Chrome and Firefox web browsers..
Safari browser make a choppy audio after few seconds from start streaming. When I switched off audio compression audio is very good quality but network usage of course much more higher.
On Chrome & Firefox is ok even with compressed audio but only in local network. When I want to connect outside of LAN even Chrome & Firefox can't resolve this situation. I do not have any Windows or Android device that I can't check on the other systems.
Is it a problem with PCM codec or with Apple OS ? .

jketterl, 2022-09-05:
Safari is not currently supported due to the fact that I don't have any Apple hardware I can do development work on. This is probably the same as reported here: https://github.com/jketterl/openwebrx/issues/303 - This won't be fixed unless we're lucky and a capable developer who has access to the right hardware happens to find the time to fix it.

I would like to know what's going on with Firefox and Chrome though. They normally work fine on any connection if the bandwidth is sufficient. Is your upstream to the internet sufficient? Can you try other receivers running version 1.2.0 from https://www.receiverbook.de/, please?

Your screenshots show that in both cases the audio playback is approximately at 48kHz - this should be fine for a soundcard running at 48kHz (please check the log window if this matches your sample rate).

jketterl, 2022-09-05:
Also: 10MS/s sample rate is definitely too much for a Raspberry Pi, so I'd recommend to at least try running a lower sample rate, e.g. 2MS/s.

Terminator0007, 2022-09-05:
Hi Jakob. 
90% of my profiles working with 2Mb/s or 5Mb/s and also I have the same situation. I have checked 32 and 64 bit debian but the same result with audio. I builded owrx also on OrangePi 4 with 6 core CPU . Next week I will try to build Openwebrx on Odroid N2+. Until today I checked these devices: owrx on RPi4B, OrangePi 4, NanoPi Neo3 with SDR's RSPduo, RSP1A, RTL-SDR and chinese clone of RSP1 also as container on Docker in Synology DS718+ with all digital voice modes. In post #303 you mentioned about mobile browsers from Apple devices but choppy audio is also on a Mac mini M1 working as stationary computer. You can install Safari dedicated to Windows OS that you can try to check what is going on. To sum up  I would like to thank you Jakob for a very good job because Openwebrx is the only one and the best one now. If this problem with audio will be repaired I will love owrx much more :-)

jketterl, 2022-09-05:
Safari for Windows has been discontinued years ago. Besides, I don't have an installation of Windows to work on either, so I'm afraid this is a dead end.

Please use one of the supported browsers (Chrome and Firefox) and refer to the groups.io mailing list for configuration support.

closed 2022-09-05 by jketterl