jketterl/openwebrx #320: Duplicate Bookmarks on same Frequency after imports

yah00078, 2022-09-22:
Hello,
First of all, thank you for all work you've done !

Using multiples PC, I regularly import locals Bookmarks to server. 

I found that every import re-add already saved frequencies, generating many duplicates entries. 

Can change the behavior ? 

Thank you !

jketterl, 2022-09-22:
Please use the issue template for feature requests.

The definition of a "duplicate" is a bit of an issue here. Checking for the frequency alone seems insufficient since frequency sharing is a thing, checking for the name in addition would mean that any edit to the name would make the problem re-appear.

I'd say the recommendation would be to delete any local bookmarks once the import has been successful, however there's the problem that the local bookmarks are currently hidden behind the server side ones. Maybe this could be an option in the import dialogue.

Another possible solution would be to only check the stations you'd actually like to import in the import dialogue.

yah00078, 2022-10-17:
Ok thank you

closed 2022-10-17 by yah00078