jketterl/openwebrx #322: On map, hyperlinked callsigns from APRS result in error

ofadam, 2022-10-03:
Due to APRS SSIDs, the hyperlinked callsigns on the map usually result in an error from qrzcq.com.

Maybe there's a way to strip that suffix out (ie, KD9NYE-10 looks up just KD9NYE, ignoring anything with "-" onward?)

(This occurs on any install from any branch of openwebrx.)

ofadam, 2022-10-03:
Updated.

jketterl, 2022-10-03:
Thank you. Tagging @luarvique since this was one of his additions.

luarvique, 2022-10-03:
Please, see this pull request:

https://github.com/jketterl/openwebrx/pull/318

It strips the station ID from APRS callsigns, as well as computing and displaying the distance between clicked beacons/squares and the QTH.

jketterl, 2022-10-04:
Well hidden.

I'm currently looking at the APRS decoder, and it seems like the information is actually separate in the AX25 header, but is merged into a single string right in the beginning. It may be more difficult, but I believe that processing the data correctly may be the better way to go. Let's see...

ofadam, 2022-10-04:
It looks great now and no longer causes errors when doing a callsign lookup.

jketterl, 2022-10-04:
Yes, I'm aware. This actually goes deeper, it is part of the AX.25 protocol, which is the base of packet radio in general.

The header format reserves 6 bytes for the callsign itself in ASCII, while the SSID is stored as a 4-bit integer in the 7th byte of the address fields. These two things need to be parsed separately anyway, so carrying this information in a structured way and rendering it at a later point would be my preferred way of solving this.

jketterl, 2023-01-05:
A late update on this: I changed the internal data model to separate callsigns and SSIDs and updated the rendering now. I didn't see any new issues popping up, so I'll close this as resolved for now. If there is still problems, please don't hesitate to open a new issue.

closed 2023-01-05 by jketterl