jketterl/openwebrx #323: Background / Fireground SSTV Decoder

luarvique, 2022-10-05:
**Feature description**
Adding SSTV decoder, similar to existing FT8, etc. decoders, both for background and foreground user. When running in the background, save decoded images to a /tmp subfolder, deleting or compressing older images. Provide web browser access to the image folder, either as a list of files (simple), or some gallery-like UI (more complex).

**Target audience**
Hams, SWLs, and DXers - all of them.

There is a command-line SSTV decoder written in Python and available here:

https://spacecruft.org/spacecruft/sstvdec

It can either be integrated into OpenWebRX as-is, or used as a sample for writing a proper decoder in C/C++.

jketterl, 2022-10-05:
Yeah this has been on the project board pretty much since I started working on the code. Please check for duplicates.

I'd say that storing temporary files should be avoided. The preferred communication is still the websocket. I also believe that implementation details should not be part of a feature request in the first place.

luarvique, 2022-10-05:
How users find the decoded images is not entirely an implementation detail. Also, the stored SSTV images are **transient** rather than temporary in this case. While there is no reason to keep them around for prolonged time (hence suggestion to use /tmp), users should still be able to accumulate and access them in some way. Websocket will not be sufficient for this purpose.

jketterl, 2022-10-05:
Background decoding may provide the data, but I don't see it as the task of OpenWebRX to store and present past transmissions. As such, the gallery feature is probably unrealistic. Long-term analysis and processing should be something that should be passed to other applications.

The display in the receiver client should probably not show more than the image currently being decoded, and maybe keep displaying the last image while there is no new transmission detected.

closed 2022-12-01 by luarvique