jketterl/openwebrx #327: container starts getting stuck at "owrx.config.migration - DEBUG - migrating config from version 1"

lesca, 2022-10-31:
**Describe the bug**
container starts getting stuck at
> owrx.config.migration - DEBUG - migrating config from version 1

I waited for 5 mins, no move on.

**To Reproduce**

```
$ sudo docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-settings:/var/lib/openwebrx jketterl/openwebrx:stable
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-10-31 05:53:17,274 - owrx.__main__ - INFO - OpenWebRX version v1.2.1 starting up...
2022-10-31 05:53:17,280 - owrx.config.migration - DEBUG - migrating config from version 1
```

**Expected behavior**
continue to start and be ready to access web.

**Installation method**
It's a docker.io environment running on NanoPi M4. OS info
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
```

**Versions**
jketterl/openwebrx:stable

**Log messages**
see above

**Additional context**
it's my first time to run openwebrx. I searched the history issues about "docker" but find no useful information. Any help is appreciated.

jketterl, 2022-10-31:
Have you tried using it? The output you have posted is basically as expected, there is no explicit "I am ready" message to wait for.

lesca, 2022-11-01:
Hello @jketterl ,
When the log comes to here, I tried access the http://ip:8037 nothing appears, then I tested the port 8037 with `telnet`, it indicated the port closed.
> 2022-10-31 05:53:17,280 - owrx.config.migration - DEBUG - migrating config from version 1

Then, I used `nc -l 8037` to open the port, `telnet` indicated it's open -- i.e. no firewall block rules prevent the access.

jketterl, 2022-11-01:
I'd say there's something wrong with the port forwarding in your docker setup. The `-p 8073:8073` argument to `docker run` should always allocate the local port, no matter what the state of the application in the container is. It doesn't even matter if the application actually actually uses the specified port or not. This should result in `nc -l 8073` returning an error (`nc: Address already in use` in my experiments).

lesca, 2022-11-01:
Hello @jketterl ,
Yes, running `nc -l 8073` with container running get error as you said. I of course stopped the container to run the command, so that I can exclude the firewall issues.

The problem here is that the startup process won't continue. it sutcks at `migration config from version 1`, even if I waited 10 or more minutes. 

You can give me some tips so I can check it further. I notice the script should stuck in this file:
https://github.com/jketterl/openwebrx/blob/develop/owrx/config/migration.py

but have no idea how to work around it or fix it.

jketterl, 2022-11-01:
As I already pointed, the line `migrating config from version 1` is the last line when the container is started up (at least as long as it still has the default configuration) and ready to serve connections, it is not an indication that the migration (or any other part of the program) is stuck. This is the expected output when the process is waiting for a connection.

Given that there's no indication of a bug, I would like to close this and move the discussion to the mailing list.

As far as the firewall analysis goes, you should keep in mind that running `nc -l` locally is different from running a docker container since docker will route traffic through separate virtual interfaces. There is a lot to go wrong here, and I'm not familiar with your particular environment. All I can recommend is to repeat the test with a comparable setup, i.e. a different type of server running in a container, with the same port forwarding in place.

Also, if you have made changes to `openwebrx.conf` please keep in mind that the port numbers in there need to line up with the port forwarding.

lesca, 2022-11-01:
Hello @jketterl ,
I don't think so. As the document on the docker hub mentioned:
> This will create a docker volume named openwebrx-settings and mount it inside the container. The startup scripts will copy the contained example config onto that volume on the first startup.

I see nothing in the `openwebrx-settings` volume.

lesca, 2022-11-01:
Hello @jketterl ,
I finally make it work by changing the default port to 5555. Below is my settings to run the container:

```docker-compose
version: '3.2'

services:
  openwebrx:
    image: jketterl/openwebrx:stable
    ports:
      - "5555:5555/tcp"
    container_name: openwebrx
    restart: always
    volumes:
      - ./openwebrx:/var/lib/openwebrx
      - ./etc/:/etc/openwebrx
    devices:
      - /dev/bus/usb:/dev/bus/usb
    tmpfs:
      - /tmp/openwebrx

```

I am sure there is no port forwarding issues with my docker environment, I tested it with a simple nginx container, everything works fine on port 80 and port 8073.

jketterl, 2022-11-06:
Well, your guess is as good as mine as to what really causes this. A proper analysis of the problem may reveal interesting information.

As far as the software is concerned, the two ports are equally sufficient to perform the task at hand, so I don't see why any change on the side of OpenWebRX should be necessary. Closing this.

closed 2022-11-06 by jketterl