jketterl/openwebrx #333: Unable to run the last stable docker image on a raspberry pi 4

fustinoni-net, 2022-12-03:
**Describe the bug**
Last docker stable image doesn't start on a raspberry pi. The last image I'm able to run is: jketterl/openwebrx:1.1.0-armv7l with:

`
docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-settings:/var/lib/openwebrx jketterl/openwebrx:1.1.0-armv7l
`

**To Reproduce**
Install the software using the wiki instruction:
`
docker volume create openwebrx-settings
docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-settings:/var/lib/openwebrx jketterl/openwebrx:stable

`

**Expected behavior**
Working like previous images

**Installation method**
How did you install OpenWebRX? (Raspberry Pi SD card image, Debian / Ubuntu packages, Docker image, manually?)
Docker with:
`docker volume create openwebrx-settings
docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-settings:/var/lib/openwebrx jketterl/openwebrx:stable`

**Versions**
What version of OpenWebRX are you running? (Check on startup, or see `owrx/version.py`. If a `-dev` version is used, ideally state the commit the issue is appearing on)
jketterl/openwebrx:stable pulled the latest

**Log messages**
Are there any relevant messages relating to the bug in the output / log of OpenWebRX? (On most installations, the log should be available using the command `sudo journalctl -u openwebrx`)
The docker output:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Fatal Python error: init_interp_main: can't initialize time
Python runtime state: core initialized
PermissionError: [Errno 1] Operation not permitted

Current thread 0xb6faf520 (most recent call first):
<no Python frame>
[cmd] /opt/openwebrx/docker/scripts/run.sh exited 1
Hello, I'm the codecserver.
now scanning for modules...
registering new driver: "ambe3k"
loading devices from configuration...
auto-detecing devices...
scanning for "ambe3k" devices...
device scan complete.
received signal: 15
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] syncing disks.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.

**Additional context**
Add any other context about the problem here.

jketterl, 2022-12-03:
First off, please clarify what version you are trying to run (`stable` or `1.1.0-armv7l`). Second, please don't use the platform-specific tags (ending in `-armv7l`), just use the regular tag instead (i.e. `1.1.0`). Docker can decide the platform itself.

Other than that, please provide information about your host OS and what changes you made since you had the previous image running. Initial suspicion is that this is one more case where the host OS is outdated and blocks calls to kernel functions because it simply doesn't know them.

fustinoni-net, 2022-12-03:
Hi,
thank you for your quick replay. I use this occasion for thank you, and the previous author, for this wonderful software.
Now, I'm using a raspberry pi 4 with Raspberry pi os:
Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
If I run the command:

`docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-settings:/var/lib/openwebrx jketterl/openwebrx:1.1.0`

I can have the software runnig.

If I run the command:

`docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-settings:/var/lib/openwebrx jketterl/openwebrx:1.2.0`
or
`docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-settings:/var/lib/openwebrx jketterl/openwebrx:stable`

the software doesn't start and I get the error:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Fatal Python error: init_interp_main: can't initialize time
Python runtime state: core initialized
PermissionError: [Errno 1] Operation not permitted

Current thread 0xb6fc0520 (most recent call first):
<no Python frame>
[cmd] /opt/openwebrx/docker/scripts/run.sh exited 1
received signal: 15
Hello, I'm the codecserver.
now scanning for modules...
registering new driver: "ambe3k"
loading devices from configuration...
auto-detecing devices...
scanning for "ambe3k" devices...
device scan complete.
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] syncing disks.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.

Unfortunately I'm not running the latest release of the Raspberry pi os. Can this be the problem?

jketterl, 2022-12-03:
> Unfortunately I'm not running the latest release of the Raspberry pi os. Can this be the problem?

Likely. Kernel 5.10 indicates that you're probably running Debian Buster. The docker images have been moved forward to use Debian Bullseye (Kernel 5.15) with the latest release (1.2) of OpenWebRX.

You don't necessarily need to upgrade, though. If you know what you're doing, you can upgrade `libseccomp` to a newer version (from backports or sid / unstable).

This is unfortunately an inherent problem of the docker infrastructure.

fustinoni-net, 2022-12-03:
I'm trying with a new installation:

Linux rpi-64 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux

Now the application works with 
`docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-settings:/var/lib/openwebrx jketterl/openwebrx:stable` 
but I get:
...
...
...
Found 1 device(s):
  0:  #, ͝, SN: 

usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
device could not be opened
Connector::open() failed
...
...
...

On the host OS I have install all the necessary files and the rtl-sdr.rules  too.
Running rtl_test:
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
lost at least 32 bytes
Signal caught, exiting!

User cancel, exiting...
Samples per million lost (minimum): 2

can you please point me to a solution also of this problem?
Thank you :-)

jketterl, 2022-12-03:
The only idea I have about that is that you may need to reboot or re-plug your device to trigger a re-evaluation of udev rules since new rule files will not immediately take effect.

Other than that, I am going to close this. The initial issue had nothing to do with OpenWebRX itself, and has been resolved. If you need further support, please visit the mailing list.

closed 2022-12-03 by jketterl