jketterl/openwebrx #337: [Feature Request] Installation Package as Service for ArozOS

Secarian, 2023-02-09:
Dear OpenWebRX Development Team,

I hope this message finds you well. I am writing to request a new feature for OpenWebRX, a software defined radio receiver application.

I have searched the issue tracker and the OpenWebRX github project, and I have not found a similar request for this feature.

**Feature description**
I would like to request the creation of an installation package for OpenWebRX as a service for ArozOS. This package would greatly simplify the installation process for users, especially for those with limited computer experience, as ArozOS greatly simplifies the process of installing packages. Additionally, the installation of OpenWebRX as a service on ArozOS would provide a number of benefits for users, including easier accessibility and lower entry barriers, especially for Raspberry Pi users.

**Target audience**
This feature would be of interest to a wide range of users, including hams, SWLs, DXers, and those with limited computer experience.

For your reference, I have included the following link, which outlines how to integrate web apps and services into ArozOS: https://github.com/aroz-online/WebApp-and-Subservice-Index

Thank you for your time and any help you can offer.

jketterl, 2023-02-18:
OK I've had a quick look at this. Contrary to the name, ArozOS is not really a fully-fledged OS, but instead just the name for the service that runs the web desktop. That means that ArozOS is not a suitable packaging platform for OpenWebRX.

The good news is that ArozOS seems to share at least one host distribution with OpenWebRX, namely Debian. That means that the OpenWebRX package repository can be used to install OpenWebRX alongside ArozOS.

Further integration with the desktop integration seems to be somewhat unclear. There seems to be some way to implement web application using the ArozOs infrastructure, but since OpenWebRX comes with its own HTTP stack this does not apply. I could not find any way to integrate an existing application into the desktop.

Does the web desktop feature a browser? If so, it should be possible to navigate to the OpenWebRX receiver client just as on normal desktop environments.

jketterl, 2023-03-17:
Is there any additional input on how this should be integrated? The available documentation does not seem to cover the use case at hand. If there's not way to integrate an existing application, I'm afraid this is a dead end.

closed 2023-05-18 by jketterl