jketterl/openwebrx #341: Messages sent to pskreporter appear to have large gaps in the sequence numbers

pjsg, 2023-02-25:
**Describe the bug**
The messages, as they arrive at pskreporter have sequence numbers which are 45 apart. This causes problems as the system thinks that there are a lot of missing packets.

**To Reproduce**
It appears that it happens all the time.

**Expected behavior**
Consecutive packets should have sequence numbers that increase by 1.

**Installation method**
N/A

**Versions**
OpenWebRX v1.3.0-dev

**Log messages**
SR4DON@549728 (xxx.yyy.1.92:48285 - OpenWebRX v1.3.0-dev): Missing 44 packets between 549343 and 549388 (549728,549683,549638,549613,549568,549523,549478,549433,549388,549343)

The numbers at the end of the line are the 10 last sequence numbers received. They appear to be exactly 45 apart (though, that would mean that they don't start at zero or one)

**Additional information**
Pskreporter supports frequencies above 4GHz -- you can send 5 byte frequencies which gets you up to 1THz. Just change the template to indicate that you are sending 5 byte frequencies. Check https://pskreporter.info/pskdev.html for the details.

jketterl, 2023-02-26:
This is indeed a change I made when updating the frequency field to 5 bytes. Funny enough, this was mostly caused by the fact that wireshark was complaining about a divergence between expected and actual sequence numbers. Under the assumption that wireshark knows more about IPFIX than i do, I changed the increment to the number of "flows" (reported spots) contained in a packet, since that seems to be what wireshark expects.

The increment of 45 results from most packets containing 45 reports after applying a byte-size limit. This is just the most common cause and may vary, depending on the length of the individual reports.

I can restore the original behavior, but I would like to point out that the definition of the sequence number in [RFC5101](https://www.rfc-editor.org/rfc/rfc5101#section-3.1) refers to "data records" and not packets, so I believe the current implementation is actually closer to the standard. The pskreporter dev guide does not further specify how exactly the sequence number should work, so maybe this should be another addition, at least as long as it's non-standard.

pjsg, 2023-02-26:
You are exactly right -- I'll fix the code on my end....

jketterl, 2023-03-17:
Alright, I guess that means I can close this on here.

Feel free to report other issues though. It's hard to see what's going on on your end, the protocol is kind of "fire and forget".