jketterl/openwebrx #345: DC blocking does not seem to work on AM

godarklight, 2023-04-13:
DC blocking does not appear to work on AM modulation
The module does appear to be loaded: https://github.com/jketterl/openwebrx/blob/develop/csdr/chain/analog.py#L13
But I am not sure if this has any taps? https://github.com/jketterl/pycsdr/blob/develop/src/dcblock.cpp#L15-L18

**To Reproduce**
Tune to an non-local/weaker AM station. I am using WWV as a test here in Australia.

**Expected behavior**
DC blocking being applied to prevent the unpleasant "clicking"

**Installation method**
Installed via git clone and then dpkg-buildpackage, dpkg -i.

**Versions**
I'm using the latest development commit.

**Log messages**
N/A

**Additional context**
I converted the mp3 to mp4 for github support.
https://user-images.githubusercontent.com/4347216/231757069-6dfcd50a-c79e-4e2f-9aae-5a2ef9aba78a.mp4

jketterl, 2023-04-13:
From my understanding, clicking is not caused by a DC offset. It could be caused by a *change* in DC offset, but that's beyond what a DC offset filter can fix.

For reference, the actual implementation of the DC blocking resides here: https://github.com/jketterl/csdr/blob/develop/src/lib/dcblock.cpp

it looks like the computation is done in blocks of 1024 samples, so if the DC offset changes between blocks, you would indeed get clicks inbetween. There's still the question as to why the offset would change. The reference link for that algorithm however is dead by now, and I did not write that code myself, so I'll have to familiarize with the topic before I can offer any fixes.

jketterl, 2023-04-13:
OK I've had a bit of a brain-block there... we're talking about AM, so the DC component actually directly relates to changes in signal strength, so yeah, this really would be a problem with fluctuating signals.

I've tried using the [DC blocker implementation from digiham](https://github.com/jketterl/digiham/blob/develop/src/dc_block/dc_block.cpp) as a replacement, and it seems to do a better job. The problem is that I don't want to have AM demod depend on digiham, so the code will need to be ported to csdr.

godarklight, 2023-04-13:
Thanks for the quick reply. I may have jumped the gun a bit and did more testing, if I remember correctly the older version that called CSDR processes didn't have the bug, nor does the csdr chain listed on the repo (example here):
`rtl_sdr -s 2400000 -f 145000000 -g 20 - | csdr convert_u8_f | csdr shift_addition_cc `python -c "print float(145000000-144400000)/2400000"` | csdr fir_decimate_cc 50 0.005 HAMMING | csdr amdemod_cf | csdr fastdcblock_ff | csdr agc_ff | csdr limit_ff | csdr convert_f_s16 | mplayer -cache 1024 -quiet -rawaudio samplesize=2:channels=1:rate=48000 -demuxer rawaudio -`

The signal does look extremely strange - I'll do more testing after work to confirm if the CSDR chain works.
[image]

godarklight, 2023-04-13:
Apparently I'm not working today, the CSDR chain works perfectly. I've got the time so I'll see if I can figure out what's going on.

jketterl, 2023-04-13:
As far as my experimentation goes, it only affects certain stations, so this may also be related to something on the transmitting end.

godarklight, 2023-04-13:
I'm also seeing it affect certain stations as well, but many of them. Removing the DC block does appear to make it go away but that's obviously not a solution - I'm starting to suspect the hyper aggressive AGC, I'll try pulling in the AGC from OpenWebRX_E and see if that cleans things up - I know their SSB sounds a lot better with it.

I initially thought there was something like an AGC/DC block flip from the port but it appears to be in the correct order...

godarklight, 2023-04-14:
It seems like you need a mid-strength station that is getting affected by some QSB to trigger it. Local stations do not cause it, nor do extremely weak stations.

I replaced the csdr/lib/dcblock.cpp file with:

`void DcBlock::process(float* input, float* output) {
    float avg = 0.0;
    size_t input_size = getLength();
    for(int i = 0; i < input_size; i++) {
        avg += input[i];
    }
    avg /= input_size;

    float avgdiff = avg - last_dc_level;
    last_dc_level = avg;

    //DC removal level will change lineraly from last_dc_level to avg.
    float firstDC = 0.0;
    float midDC = 0.0;
    float endDC = 0.0;
    for(int i = 0; i < input_size; i++) {
        float dc_removal_level = last_dc_level + avgdiff * ((float) i / input_size);
        output[i] = input[i] - dc_removal_level;
        if (i == 0)
        {
          firstDC = dc_removal_level;
        }
        if (i ==  input_size / 2)
        {
          midDC = dc_removal_level;
        }
        endDC = dc_removal_level;
    }
    std::cerr << "DC " << firstDC << ":" << midDC << ":" << endDC << std::endl;

}`

And got this result:
Apr 14 09:53:51 chrism5owrx openwebrx[37601]: DC 0.000217727:0.000212542:0.000207367
Apr 14 09:53:51 chrism5owrx openwebrx[37601]: DC 0.000249833:0.000265886:0.000281908
Apr 14 09:53:51 chrism5owrx openwebrx[37601]: DC 0.000429705:0.000519641:0.000609401
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000585158:0.000662884:0.000740459
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.00047274:0.000416532:0.000360433
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000254948:0.000146053:3.73693e-05
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000314142:0.000343739:0.000373278
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000401214:0.00044475:0.000488201
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000463984:0.000495369:0.000526692
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000478764:0.000486153:0.000493529
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000569064:0.000614214:0.000659276
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000697354:0.000761498:0.000825518
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000715813:0.000725043:0.000734255
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000676955:0.000657525:0.000638134
Apr 14 09:53:52 chrism5owrx openwebrx[37601]: DC 0.000727711:0.000753089:0.000778418
Apr 14 09:53:53 chrism5owrx openwebrx[37601]: DC 0.000873029:0.000945688:0.00101821
Apr 14 09:53:53 chrism5owrx openwebrx[37601]: DC 0.000880879:0.000884804:0.000888721
Apr 14 09:53:53 chrism5owrx openwebrx[37601]: DC 0.000789913:0.00074443:0.000699036
Apr 14 09:53:53 chrism5owrx openwebrx[37601]: DC 0.000793315:0.000795016:0.000796714
Apr 14 09:53:53 chrism5owrx openwebrx[37601]: DC 0.000978992:0.00107183:0.00116449
Apr 14 09:53:53 chrism5owrx openwebrx[37601]: DC 0.00112373:0.0011961:0.00126833

So something extremely suspicious is going on with the linear scaling of the DC offset... I have never done python-to-c++ interfacing however so not entirely sure what the cause could be. I would expect the first and end result of the last line to be the same.

godarklight, 2023-04-14:
I believe this line should be moved after the for loop, I think it's supposed to slowly interpolate over the range, not jump to the new value and keep interpolating.

https://github.com/jketterl/csdr/blob/develop/src/lib/dcblock.cpp#L50

godarklight, 2023-04-17:
This is not an openwebrx issue but a csdr issue, I've opened a PR over there - Feel free to close this at any time after deciding what to do to the PR.

jketterl, 2023-04-18:
Thank you for the pull request, I'll have a look as soon as I find the time.

Could you still do me a little favor and try out the digiham DcBlock? It is a drop-in replacement in python, so if you have digiham + pydigiham installed, a simple (temporary) change to the imports should be all that is needed to switch.

The reason I'm asking is that there is currently two DC block implementations present, and I'd like to take the occasion and get rid of one of them. To me, it seems like the one in digiham would be the better one, but so far I've only been able to reproduce the original problem on one single station.

godarklight, 2023-04-18:
From what I can see it's mostly noticeable on signals with voice or stations with some pretty bad QSB. I did attempt to test the change and heard no difference to my ear, but the sun is coming up and I'm going to lose all of the DX shortwave stations very shortly.

It's definitely possible that digiham's implementation is better, it looks like a normal 2 point IIR high-pass filter, but for a fair test I'd need to record IQ and play it back as an rtl_tcp source.

closed 2023-04-21 by godarklight

jketterl, 2023-04-21:
Thank you for testing. I'll see what I can come up with. Also thanks for reporting back about the fix, glad it's resolved now.