jketterl/openwebrx #348: codec2 url has changed

jubamo, 2023-07-22:
In "openwebrx/docker/scripts/install-dependencies.sh"

Changed url of:
 git clone https://github.com/drowe67/codec2.git
cd codec2

to:
git clone https://github.com/drowe67/codec2-dev.git
cd codec2-dev

jketterl, 2023-07-22:
Any reference or reason for this? if I look at the readme of the `codec2-dev` repository, it says:

> Note This repo contains deprecated code, the latest code can be found at https://github.com/drowe67/codec2

So at least to me, it seems that that change would be counterproductive.

jubamo, 2023-07-22:
In my docker image I use the newer version and have no problems.

The reason for the url change is none other than to keep the same version
that was being used.

Of the two options that exist, keep the version or use a current one, I had
chosen to keep it to avoid any incompatibility, only that.

Juan.

jketterl, 2023-07-22:
I'm sorry, but the fact that it works for you is not an indication to switch. Since the suggested repository is deprecated, I don't consider it to be a reasonable choice.

The build with the current repository should work fine, and the fact that the version is pinned to a specific git commit should ensure that it is also reproducible.

closed 2023-07-22 by jketterl

jubamo, 2023-07-22:
Sorry, I didn't explain myself well, my suggestion to change the url is to
keep the same pinned version of a commit that is being used, if nothing is
changed, the docker image cannot be generated

jketterl, 2023-08-04:
Perfect example of why you should always explain a problem, and not just give instructions to random strangers. The issue template would probably have helped with that.