jketterl/openwebrx #350: POCSAG : message window do not handle < ; : char

psa-jforestier, 2023-08-11:
**Describe the bug**
When a message received by the POCSAG decoder contains a `<` char followed by an other char (like `<H` or `<:`, the message is trucated in the message window. Problem similar with some other chars like `;` and `:`.

**To Reproduce**
The POCSAG message is `kK<:HJHZI"I*I2I:IJJ"J*JBHI` but the message window display only `kK`.
(this kind of message seems weird but are valid : used as a time sync mecanism)

**Expected behavior**
The text message must be printed on the message window as received.

**Installation method**
Use an online web-sdr.

**Versions**
OpenWebRX version: v1.1.0

**Log messages**
Nop.
, 
**Additional context**
There is probably some other bugs related to message encoding, because when I searched for a workaround and tried to patch the JS function `PocsagMessagePanel.prototype.pushMessage`, I saw strange output when doing `console.log(msg)` when the message contains `:` or `;` . 

A quick fix to the `<` problem is to add a ".toHtmlEntities()" (with proper JS implementation) to the "msg.message" here : https://github.com/jketterl/openwebrx/blob/567369969607f896c5f795a7396b085c567cb8a9/htdocs/lib/MessagePanel.js#L260
But it did not solve the : and ; problem.

jketterl, 2023-08-11:
The problems with `;` and `:` are probably down to the way the messages are passed in the old version (1.1.0). This has been reworked in version 1.2, so please update.

The escaping of `<` and `>` as html special chars is something i can look into.

jketterl, 2023-10-12:
This has now been backported and released in version 1.2.2. Closing.

closed 2023-10-12 by jketterl