jketterl/openwebrx #352: APRS: Some symbols point EAST, others point NORTH (or nowhere)

luarvique, 2023-08-29:
**Describe the bug**
Current code assumes all APRS symbols pointing EAST, making NORTH-pointing symbols point in the wrong direction.

**To Reproduce**
Just find an airplane.

**Expected behavior**
North-heading airplane should point north.

**Installation method**
Packages

**Versions**
OWRX 1.2.1 and derived (OWRX+)

**Log messages**
Nope

**Additional context**
See proposed solution in the following message.

luarvique, 2023-08-29:
Here is the proposed solution, which essentially filters out east-pointing symbols for the special rotation algorithm:

```
    const FACING_EAST = "(*<=>CFPUXYabefghjkpsuv[";

    if (!this.course) {
        div.style.transform = null;
    } else if (this.symbol && !FACING_EAST.includes(this.symbol.symbol)) {
        // Airplanes and other symbols point up (to the north)
        div.style.transform = 'rotate(' + this.course + 'deg)';
    } else if (this.course > 180) {
        // Vehicles and vessels point right (to the east)
        div.style.transform = 'scalex(-1) rotate(' + (270 - this.course) + 'deg)'
    } else {
        // Vehicles and vessels point right (to the east)
        div.style.transform = 'rotate(' + (this.course - 90) + 'deg)';
    }
```

jketterl, 2023-08-30:
Thanks. I worked this out a bit more to also include the symbol table, see commit above.

luarvique, 2023-08-30:
Acknowledged. I skipped on the table selection, since most of the alt table symbols looked ok rotating either way. But this look more complete.

closed 2023-09-12 by jketterl