jketterl/openwebrx #361: APRS reporting doesn't work on 1.2.2 version

nmuntyanov, 2023-11-05:
**Describe the bug**
I've have installed Openwebrx + direwolf, but aprs packets can't be decoded and reported

**To Reproduce**
```
ssh user@host
sudo -i
wget -O - https://repo.openwebrx.de/debian/key.gpg.txt | gpg --dearmor -o /usr/share/keyrings/openwebrx.gpg
echo "deb [signed-by=/usr/share/keyrings/openwebrx.gpg] https://repo.openwebrx.de/ubuntu/ jammy main" > /etc/apt/sources.list.d/openwebrx.list
apt-get update
apt-get install openwebrx
apt-get install direwolf
systemctl restart openwebrx
```

**Expected behavior**
aprs packets should be decoded and reported on APRS-IS network

**Installation method**
Installation source: https://www.openwebrx.de/download/ubuntu.php stable version
HW: OrangePi zero 2w
OS: Ubuntu 22.04.3 LTS

**Versions**
1.2.2

**Log messages**
```
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]: Exception in thread Thread-327:
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]: Traceback (most recent call last):
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:   File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:     self.run()
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:   File "/usr/lib/python3.10/threading.py", line 1378, in run
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:     self.function(*self.args, **self.kwargs)
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:   File "/usr/lib/python3/dist-packages/owrx/service/__init__.py", line 167, in updateServices
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:     self.services.append(self.setupService(dial["mode"], dial["frequency"], self.source))
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:   File "/usr/lib/python3/dist-packages/owrx/service/__init__.py", line 250, in setupService
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:     secondaryDemod = self._getSecondaryDemodulator(modeObject.modulation)
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:   File "/usr/lib/python3/dist-packages/owrx/service/__init__.py", line 292, in _getSecondaryDemodulator
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:     from csdr.chain.digimodes import PacketDemodulator
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:   File "/usr/lib/python3/dist-packages/csdr/chain/digimodes.py", line 8, in <module>
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]:     from digiham.modules import FskDemodulator, PocsagDecoder
syslog:Nov  5 10:27:50 orangepizero2w openwebrx[2343]: ModuleNotFoundError: No module named 'digiham'
```

`apt-get install digiham` has no effect

jketterl, 2023-11-05:
Yes, the dependencies are not tracking this correctly. This has been fixed in the development version.

As a workaround, please install pydigiham (`apt-get install python3-digiham`).

Related thread: https://groups.io/g/openwebrx/topic/92109805

nmuntyanov, 2023-11-05:
Got it, thank you.

closed 2023-11-05 by nmuntyanov