jketterl/openwebrx #363: RDS decoding

oh2lak, 2023-11-13:
**Feature description**
Add RDS decoding to WFM demodulation

**Target audience**
RDS decoding would benefit everyday WFM broadcast band listener but would be especially helpful for FMDX use to identify stations

An CLI RDS decoder exists at https://github.com/windytan/redsea which claims that it can be used with CSDR

jketterl, 2024-01-20:
Implemented in the development version now. Give it a spin and let me know what you think.

KOTYA8, 2024-04-09:
Hello, thanks for adding RDS, but is it possible to make sure that when the RT+ function is activated, the main RT line does not disappear? In order to see what is written in RT.

jketterl, 2024-04-09:
From what I understand, the internal implementation of RT+ actually uses the "legacy" RT text / line data, and just applies a tagging mechanism. That means that ideally, RT+ and RT would always display the same information (worst case is that RT+ leaves out some information, but it cannot display different information). As such, I didn't see the need to show them both at the same time. I was also considering how an actual radio would behave, and from what I've been able to find, they would also be displaying only one of the two systems (depending on what is available).

KOTYA8, 2024-04-09:
Yes, it looks ugly when there are two lines. But I want to watch the full version of RT. Only the song is shown and that's it. Or somehow disable RT+. Thank you.

jketterl, 2024-04-09:
Well, what I'm trying to tell you is that... if RT+ displays the song, RT would also display the song.

KOTYA8, 2024-04-09:
But it turns out that when RT changes and it does not have an RT+ program, it completely disappears and only PS remains.

jketterl, 2024-04-09:
well, there's two reasons why this would happen:

- the stations transmits RT+, but with empty tags. I'm not sure why they would do that, but my expectation is that you'd see the same results on other RT+ capable receivers.
- the station does transmit tags, but they can't be displayed by OpenWebRX. There is a number of tags that are currently unimplemented because I haven't encountered them, and don't know exactly how to display them. If you know of any tags that should be displayed, please share more details.

KOTYA8, 2024-04-09:
Yes, they have other tags, there seem to be (Stationname.Short, Stationname.Long) and there is a simple RT that does not have the RT+ program, I’m just an RDS collector and wanted to know what is transmitted to RT. So how can we show RT without any problems with additional RT+ programs?

xxj737, 2025-10-16:
for example：

[image]
[image]
[image]