jketterl/openwebrx #366: Adjustable AGC

ssemkaz, 2023-11-29:
Initially, AGC is configured terribly and just spoils the sound quality, it would be nice to see a configurable AGC

jketterl, 2023-11-29:
Please use the issue template.

jketterl, 2023-11-29:
Needs clarification: What AGC are we talking about? SDR device AGC, or audio AGC?

This is probably a request that has been made many times before, I'll just leave this open since it's probably the only one here in the issue tracker. The audio AGC has been prepared to accept various parameters at runtime already, it just needs a user interface, which I have not added yet since I don't intend to overload the current receiver controls. A new UX concept and rework will be necessary before.

For future reference:

https://groups.io/g/openwebrx/topic/101714030
https://groups.io/g/openwebrx/topic/98603846
https://groups.io/g/openwebrx/topic/97955307
https://groups.io/g/openwebrx/topic/83646821
https://groups.io/g/openwebrx/topic/77469770

Probably a few more that I've missed.

ssemkaz, 2023-11-29:
I'm talking about audio AGC, judging by the answers from those topics that you sent, I realized that this setting will either not be added at all, or it will be but not soon? If it is not added at all, will the ability to manually adjust the level be added or will the speed be changed? Thanks for the answer in advance

closed 2023-11-29 by ssemkaz

jketterl, 2023-11-29:
I don't know how you're coming to the idea that it will not be added at all. I think I have explained multiple times that the challenge is to find a reasonable user interface, so once that has been sorted the corresponding AGC controls can be added. The AGC algorithm itself already allows adjusting a number of parameters and already supports different presets internally.

I don't intend to change the current defaults since they are very subjective and I consider it impossible to find a set of parameters that would work for everybody.

ssemkaz, 2023-11-29:
Then how can I do it now, if at all?