jketterl/openwebrx #368: Feature Request: Add NETSDR Support

df6dbf, 2024-01-12:
Hello Jakob,
i have here a NetSDR compatible SDR Receiver "Andrus MK1.5". Is it possible to add NetSDR Protocol as used also by RFspace SDR Radios?

73 Holger, DF6DBF

jketterl, 2024-01-12:
Please use the issue template.

Seems like there's a SoapySDR module available: https://github.com/pothosware/SoapyNetSDR

How about testing? Since I don't own a compatible device, I won't be able to do a lot more than just basic integration. If that doesn't work, somebody will need to figure out the details.

Are you able to run an OpenWebRX branch if provided?

df6dbf, 2024-01-13:
Hallo Jakob,
i have tried to add the NetSDR Module but with no success. (Using here the fork openwebsdr+). But I can lend you a NetSDR compatible receiver for testing purposes if needed.

Call @ darc.de

73s Holger

jketterl, 2024-01-14:
I added a branch "netsdr" and added the basic device support there, or at least what i could deduce from the available information. If anybody is able to test it out, please report back here.

RobinCutshaw, 2024-11-18:
I have a NetSDR.  I applied your netsdr branch modifications to the current branch.  The sdcard image I loaded did not include the driver.  I cloned SoapySDR, checked out the commit back to 0.8.0 which the sdcard image uses and compiled.  I cloned the SoapyNetSDR and built it against the cloned SoapySDR and then copied libnetSDRSupport.so to /usr/local/lib/SoapySDR/modules-0.8.  The NetSDR then showed up when adding a new device.  I created that and input the IP:port for the NetSDR.  Seems to work fine.

Note that the NetSDR has its own discovery protocol.  SoapySDRUtil --probe="driver=netsdr"  uses this protocol and shows the found device.  It would be nice if this were used when adding the device to openwebrx so you didn't have to find the IP and port to enter.

Need to get the prebuilt sdcard version to build/include the NetSDR module.

Robin

jketterl, 2024-11-19:
> Note that the NetSDR has its own discovery protocol. SoapySDRUtil --probe="driver=netsdr" uses this protocol and shows the found device. It would be nice if this were used when adding the device to openwebrx so you didn't have to find the IP and port to enter.

There's no device discovery in OpenWebRX in general, so this will have to wait.

> Need to get the prebuilt sdcard version to build/include the NetSDR module.

Well, there's the eternal conundrum. You want it to be in the SD image... But I wanted to have a confirmation that it works first. It's impossible to develop software if everybody just relies on the ready-made images.

RobinCutshaw, 2024-11-29:
I guess you need more than my say so to confirm that it is working.  It is working for me given the above configuration steps.

jketterl, 2024-12-11:
That is a good start. I will merge this once I find the time to do so. It just goes to show how difficult it is to get any kind of support on this project. It would probably be easier to just buy a device, but I'm refusing since they are already filling up my space.