jketterl/openwebrx #369: RTL_SDR device not found

wipedlifepotato, 2024-01-14:
```
user@computer:~/rtl-sdr/opensdr/j/openwebrx-develop$ rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R828D tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

User cancel, exiting...
Samples per million lost (minimum): 0
Reattached kernel driver
user@computer:~/rtl-sdr/opensdr/j/openwebrx-develop$ vi openwebrx.conf 
user@computer:~/rtl-sdr/opensdr/j/openwebrx-develop$ python3 openwebrx.py 

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Jakob Ketterl, DD5JFK <dd5jfk@darc.de>
Documentation:          https://github.com/jketterl/openwebrx/wiki
Support and info:       https://groups.io/g/openwebrx

        
2024-01-14 20:45:05,344 - owrx.__main__ - INFO - OpenWebRX version v1.3.0-dev starting up...
2024-01-14 20:45:05,346 - owrx.sdr - ERROR - The SDR source type "rtl_sdr" is not available. please check the feature report for details.
2024-01-14 20:45:05,348 - owrx.sdr - ERROR - The SDR source type "airspyhf" is not available. please check the feature report for details.
2024-01-14 20:45:05,348 - owrx.sdr - ERROR - The SDR source type "sdrplay" is not available. please check the feature report for details.
2024-01-14 20:45:05,350 - owrx.__main__ - INFO - Ready to serve requests.

```

```

This receiver is currently unavailable due to technical issues.
Error Message:
No SDR Devices available
```
But that works correctly in sdrpp, old version of openwebrx, grqx etc

jketterl, 2024-01-14:
Please use the issue template.
Please use the [groups.io mailing list](https://groups.io/g/openwebrx) for support.

jketterl, 2024-01-14:
This is just not the right place. This place is reserved for issues related to issues with the software. I can already tell that this is a problem on your end, so it has no business on here.

If you can't be bothered to use the right channel for your problem, there's nothing I can do for you.

closed 2024-01-14 by jketterl

wipedlifepotato, 2024-01-14:
And what channel I have to use? I found some simillar to my problem in your forum. There is not answers. I Found in source code that there is use soapysdr. I did compiled much libraires (csdr, python27, etc) for just use openwebrx. So. I would to try docker package but i think there is will be problem with no source problem. Ehh..

jketterl, 2024-01-14:
That is exactly the kind of shit that does not go down here. If you have questions, go over to the mailing list and ask them there. If you feel like discussing, go over to the mailing list and start a thread.

wipedlifepotato, 2024-01-14:
there is trouble with gmail supports. email not going to my mailbox, even as spam . And oon my own mail server email not going. May i register my account please

jketterl, 2024-01-14:
You sort that out with groups.io first. There is a lot of members over there using gmail. Can't help you with that, sorry.

wipedlifepotato, 2024-01-14:
Like, there is some pause before sending email. Ok. 
``` Your message will be sent when it's approved by the moderators. ```
So, for now i have to wait?

wipedlifepotato, 2024-01-14:
```# SoapySDRUtil --probe="driver=rtlsdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=rtlsdr
Detached kernel driver
Found Rafael Micro R828D tuner
Reattached kernel driver
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Detached kernel driver
Found Rafael Micro R828D tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=R828D
  index=0
  origin=https://github.com/pothosware/SoapyRTLSDR

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: YES
  Time sources: sw_ticks
  Other Settings:
     * Direct Sampling - RTL-SDR Direct Sampling Mode
       [key=direct_samp, default=0, type=string, options=(0, 1, 2)]
     * Offset Tune - RTL-SDR Offset Tuning Mode
       [key=offset_tune, default=false, type=bool]
     * I/Q Swap - RTL-SDR I/Q Swap Mode
       [key=iq_swap, default=false, type=bool]
     * Digital AGC - RTL-SDR digital AGC Mode
       [key=digital_agc, default=false, type=bool]
     * Bias Tee - RTL-SDR Blog V.3 Bias-Tee Mode
       [key=biastee, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS8, CS16, CF32
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Size - Number of bytes per buffer, multiples of 512 only.
       [key=bufflen, units=bytes, default=262144, type=int]
     * Ring buffers - Number of buffers in the ring.
       [key=buffers, units=buffers, default=15, type=int]
     * Async buffers - Number of async usb buffers (advanced).
       [key=asyncBuffs, units=buffers, default=0, type=int]
  Antennas: RX
  Full gain range: [0, 49.6] dB
    TUNER gain range: [0, 49.6] dB
  Full freq range: [23.999, 1764] MHz
    RF freq range: [24, 1764] MHz
    CORR freq range: [-0.001, 0.001] MHz
  Sample rates: [0.225001, 0.3], [0.900001, 3.2] MSps
  Filter bandwidths: [0, 8] MHz

Reattached kernel driver
```

jketterl, 2024-01-14:
You're completely off topic by now. Perfect example of why I closed this issue right away.

wipedlifepotato, 2024-01-14:
> You're completely off topic by now. Perfect example of why I closed this issue right away.

There is no need for me to assign blame and read excuses for your actions. I have a task to fix a bug, preferably something that works. I need to see if it makes sense to spend time with you at all or not, as support, as a community.
Once again I am convinced that in relation to me, open source communities are aggressive...

So, if that closed, then is closed. Ok. have a nice day.

jketterl, 2024-01-14:
There is no bug here, which is exactly the reason why I'm not going to spend any additional time on this discussion. If you can't get it to work, that's not my fault.