jketterl/openwebrx #380: Does it work with the latest RTL-SDR V4?

Andrej64, 2024-04-12:
Just asking before installing. 

Thanks.

petrkr, 2024-04-12:
If you mean RTL-SDR V4 from RTL-SDR BLOG V4, black dongle. Then for me, when I used rtl-sdr blog's drivers and replaced package one, then yes it works.

But out-of-box I think still nope, at least not in release version.

jketterl, 2024-04-12:
Please use the issue template.
Please use the groups.io mailing list for support request. This question has been answered before there.

To answer the question: It depends. OpenWebRX just uses the rtl-sdr library that is available on the system / environment it runs on. That means that OpenWebRX does not explicitly support the rtl-sdr blog v4, but at the same time does not need to do so.

Not all installation methods come with their own version of the library, so compatibility cannot be guaranteed across the board. Since the API library hasn't changed, it is possible to just replace it with a newer version to make it work if necessary.

closed 2024-04-12 by jketterl