jketterl/openwebrx #384: Distorted audio on mono and 32 kHz sampled DAB stations

orryverducci, 2024-05-26:
**Describe the bug**
Tuning in to DAB stations that broadcast either mono audio or audio using a 32 kHz sample rate results in distorted audio playback. Mono stations have stuttery audio, as if samples are regularly missing. 32 kHz stations play sped up and then drop out.

**To Reproduce**
Tune in to the frequency of a DAB multiplex containing a mono or 32 kHz samples station. Enable the DAB decoder, then select a mono or 32 kHz sampled station form the program list.

**Expected behavior**
Clean audio playback, as experienced when listening to a 48 kHz stereo DAB station.

**Installation method**
Docker image on a Raspberry Pi 4.

**Versions**
Latest tagged Docker image.

**Log messages**
None

**Additional context**
Mono and 32 kHz sampled stations are common in the UK, so many stations received from the UK DAB multiplexes are affected.

jketterl, 2024-05-27:
Yeah this has been confirmed on the mailing list before. Basically, as of now the audio data coming out of dablin is unqualified, and the best (most common) option is to interpret them as 48kHz stereo. There has been work on dablin towards adding a header (https://github.com/Opendigitalradio/dablin/issues/74) but this still needs to be integrated. See also #155 for some additional information.

At least we finally have an issue now ;)

orryverducci, 2024-05-30:
Thanks for the additional context. I did search through the mailing list but somehow missed this, so that's useful to know.

It's good to see that the dablin team are making changes to provide this information.

pbriesch, 2026-09-02:
I can reproduce a related issue on OpenWebRX+ 1.2.123 running on Debian 13 (Trixie), using:

- `dablin 1.16.0`
- `python3-csdr-eti 0.0.11`
- RTL-SDR
- DAB audio rate configured to 48 kHz

In my case this affects **all services in the multiplex, including 48 kHz stereo services**, not only mono or 32 kHz services.

The ensemble and service list decode correctly, and `dablin` is actually producing audio. I verified this with `strace`:

```text
read(0, ..., 4096) = 4096
write(1, ..., 4096) = 4096
```

So the ETI → dablin part of the pipeline is working.

The relevant OpenWebRX code currently declares the output of:

```text
dablin -p -s <service-id>
```

as:

```python
Format.FLOAT
```

while `dablin -p` outputs raw signed 16-bit PCM.

With the unmodified code, the browser audio buffer stays around 0.0–0.1 seconds and no usable audio is heard.

I tested changing the DAB chain to:

```text
Format.SHORT
→ Convert(Format.SHORT, Format.FLOAT)
→ Downmix(Format.FLOAT)
```

but this caused the DAB receiver to remain stuck at "Loading..." in my installation.

The following workaround, however, works reliably:

```text
dablin -p
→ ffmpeg s16le / 48 kHz / stereo
→ f32le / 48 kHz / stereo
→ OpenWebRX
```

Specifically, replacing the dablin command with:

```python
return [
    "bash", "-c",
    "dablin -p -s {:#06x} | "
    "ffmpeg -loglevel quiet "
    "-f s16le -ar 48000 -ac 2 -i pipe:0 "
    "-f f32le -ar 48000 -ac 2 pipe:1".format(self.serviceId)
]
```

while keeping the module output as:

```python
Format.FLOAT
```

fixes the issue completely. The service list loads, the audio buffer fills normally, and audio playback is clean.

I am aware of PR #419 and the preference to use the existing CSDR conversion/resampling facilities rather than adding ffmpeg as a dependency. My test may nevertheless be useful because a straightforward `SHORT -> FLOAT` conversion in the existing DAB chain did not work in this setup, whereas explicit `s16le -> f32le` conversion does.

This may therefore be more than just the known 32 kHz / mono sample-rate issue and could also involve the PCM sample format expected from `dablin -p`.