jketterl/openwebrx #392: Nan Hz

aa5af5607, 2024-11-13:
The openwebrx won't let me enter a frequency it tells me NaN Hz how do you correct that
Robby

closed 2024-11-13 by jketterl

aa5af5607, 2024-11-13:
> Please use the issue template.
> Please use the groups.io mailing list for support requests.

Didn't get any help from nobody on the group Sent from my iPhone

jketterl, 2024-11-13:
That doesn't mean you can bring your issues here. Also, I don't remember seeing any topics that would relate to your message there.

aa5af5607, 2024-11-13:
Maybe because I'm a newbie and it hasn't been released by the moderator yet Sent from my iPhone

jketterl, 2024-11-14:
You do realize I can see when you signed up and sent your messages there? I can clearly tell that you sent your messages after you claimed that you "Didn't get any help from nobody on the group".

Please stop shitting on our community if you haven't even bothered.