jketterl/openwebrx #396: WebSocket error crashing container

matthewric, 2024-12-10:
**Describe the bug**
Crashes the container and it restarts. I get this message in the console

WebSocket connection to 'ws://192.168.0.84:8073/ws/' failed:  receiver.js:1319 

**To Reproduce**
It happens every time I run it after about 25 to 30 seconds

**Installation method**
Docker

**Versions**
V1.2.74

[error](https://github.com/user-attachments/assets/2643a8df-51af-4bca-a9f3-305163a1980f)

0xAF, 2024-12-11:
You seems to be using the docker image of OWRX+.
As the Docker Hub page clearly says:
issues with this image should be reported directly to me [LZ2SLL⁠](https://0xaf.org/about/)
or in [the chat⁠](https://t.me/openwebrx_chat)

So, Jakob, I guess you can close this, if you're not going to dive into it.

jketterl, 2024-12-11:
This is not an official version of OpenWebRX. The last stable release of OpenWebRX is 1.2.2, so you're probably using a fork.

Please see if you can reproduce the error in an official version of OpenWebRX.

Also please note that the websocket closing is just a symptom, but does not allow proper diagnosis. You will need to provide further insight into the logs before anybody will be able to work on this.

closed 2025-11-22 by jketterl