jketterl/openwebrx #399: Any idea when Raspberry Pi 5

Andrej64, 2024-12-20:
Before posting a new feature request, please check if a similar idea has already been listed
* on the issue tracker
* on the [OpenWebRX github project](https://github.com/users/jketterl/projects/1).

In the latter case, please only proceed if you have additional information about the feature, and please let us know that there's already a card there.

**Feature description**
Please describe in plain words what functionality you'd like to see in OpenWebRX, and why you think it's useful.

**Target audience**
Please let us know if you think that this feature is of particular interest for a particular group of users (e.g. hams, SWLs, DXers, ...)

Andrej64, 2024-12-20:
Any idea when Raspberry Pi 5 (500) will be supported?
I tried it and it does not boot.
thanks

jketterl, 2024-12-21:
Please use the issue template. (Hint: you didn't fill in the blanks. This is pointless.)
Please use the groups.io mailing list for general questsions.

The Pi 5 should be supported in the latest experimental builds. I have no feedback whether or not that also extends to the Pi 500.

closed 2024-12-21 by jketterl