jketterl/openwebrx #400: Add link to local homepage (/)

dstengele, 2025-02-11:
**Feature description**
Currently there is no direct link (that I am aware of) back to the homepage when you are for example browsing the received files or the map. One would think that the logo on the top left links back to `/`, but it instead links to https://www.openwebrx.de/.

While I understand that the link is useful to make people more aware of the project itself, I would rather have it link back to the local homepage.

Alternatively, an additional button back to the homepage on the top right (next to the map, files, etc.) would also work, maybe even better as some sort of "tab bar" with the current page highlighted.

**Target audience**
Everyone ;)

jketterl, 2025-03-17:
When you're talking about "homepage", do you mean the receiver page?

dstengele, 2025-03-18:
Ah, yeah, that's what I meant :) The thing at /.