jketterl/openwebrx #404: CTCSS and DCS Decoding/Squelch funcionality

hb9dww, 2025-04-11:
Many repeaters today use CTCSS or DCS, especially if there are several repeaters on the same frequency or if a multi-mode repeater is used to suppress digital transmissions during analogue reception.

It would be very helpful for radio amateurs and other users if CTCSS/DCS decoding (Decoding of CTCSS frequencies/DCS codes) and squelch functionality could be implemented in OpenWebRX, perhaps even in conjunction with the scanning functionality.

Thank you very much for the great effort you have put into extending OpenWebRX. OpenWebRX is really a cool and powerful tool!

73 de Peter HB9DWW

jketterl, 2025-04-11:
Please use the issue template.

hb9dww, 2025-04-17:
I've now used the issue template. I hope everything is now compliant.