jketterl/openwebrx #405: DMR LRRP implementation

nsilvo, 2025-04-13:
Feature description
I Listen to some DMR systems nearby, and a couple of them use LRRP for postition reporting, is it possible to grab this using digiham module and display these on the map as per other modes? 

Anything extra that can be gleaned from DMR when scoping a system is useful.

Target audience
This would be a feature for all users who listen to DMR and potentially P25

Thanks
omgf1sh

jketterl, 2025-04-14:
Is there any specification as to where in the protocol that data is encapsulated, and how? I don't believe it's part of the DMR standard.

nsilvo, 2025-04-14:
I believe it’s part of a messaging protocol but not sure of the specifics, other than LRRP is Motorola standard with companies such as Hytera doing things slightly different.

I have found a couple of dmr based libraries with reference to the decoding of the encapsulated packets, and it’s a feature within DSD+ which probably won’t help.

Once I’m at my computer I will drop a couple of links.

Thanks